arduino/arduino-language-server #75: Sketch has error diagnostics although I can compile and upload

kittaakos, 2021-03-24:
## Bug Report

### Current behavior

I have a compiler error in my sketch. I have restarted the LS, reloaded the window; it did not work.

Error:
```
No matching member function for call to 'println'

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:79:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:78:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:77:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:80:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:81:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:82:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:83:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:84:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:85:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:86:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:87:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:88:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:89:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/akos.kitta/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:90:12:
note: candidate function not viable: requires 0 arguments, but 1 was provided
```

Sketch:
```cpp
#include "other.h"
void setup() {
  bar();
  Serial.begin(9600);
}
void loop() {
  Serial.println("foo");
}
```

`Other.h`:
```cpp
void bar() { }
```

[image]

I have selected my connected Zero board:
```
~/git/arduino-ide/arduino-ide-extension/build/arduino-cli board list --format json
[
  {
    "address": "/dev/cu.URT2",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.URT1",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.Bluetooth-Incoming-Port",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.usbmodem14602",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Arduino Zero (Programming Port)",
        "FQBN": "arduino:samd:arduino_zero_edbg",
        "VID": "0x03EB",
        "PID": "0x2157"
      }
    ],
    "serial_number": "3ZXXH6CNQ4ME0E3Q69HT"
  }
]
```

[image]

### Expected behavior

### Environment

- Language Server version (commit hash):
- CLI version used (output of `arduino-cli version`):
- OS and platform:

### Additional context

ubidefeo, 2021-03-28:
@kittaakos 
this error happens on the Zero, and it might have to do with how the `Serial` macro is defined.
It might require some rework on the core level, probably @facchinm and @cmaglie know what causes it

per1234, 2022-06-23:
Closing as fixed by https://github.com/arduino/arduino-language-server/pull/92

closed 2022-06-23 by per1234