arduino/arduino-language-server #76: Runtime Error when receiving textDocument/didChange

michaeljoelphillips, 2021-03-28:
## Bug Report

### Current behavior
Any sketch that I open with the Language Server running fails during `textDocument/didChange` with the following error in the logs:
```
# inols-err.log

2021/03/27 20:55:46 IDE --> LS     CL: NOTIFICATION textDocument/didChange:
2021/03/27 20:55:46 IDE --> textDocument/didChange notif3  locked
2021/03/27 20:55:46 --> didChange(file:///home/nomad/Code/arduino/blink/blink.ino@4)
2021/03/27 20:55:46      > <nil> -> "void setup() {\n\tpinMode(12, OUTPUT);\n}\n\nvoid loop() {\n\tdigitalWrite(12, HIGH)\n\tdelay(1000);\n\n\tdigitalWrite(12, LOW);\n\tdelay(1000);\n}\n"
2021/03/27 20:55:46 IDE --> textDocument/didChange notif3  unlocked
2021/03/27 20:55:46 Panic: runtime error: invalid memory address or nil pointer dereference

goroutine 6 [running]:
runtime/debug.Stack(0xa3495f, 0x2, 0xc0003c0990)
    /usr/lib/go/src/runtime/debug/stack.go:24 +0x9f
github.com/arduino/arduino-language-server/streams.CatchAndLogPanic()
    /home/nomad/Code/arduino-language-server/streams/panics.go:14 +0xc7
panic(0x982d40, 0xe06270)
    /usr/lib/go/src/runtime/panic.go:971 +0x499
github.com/arduino/arduino-language-server/handler/textutils.ApplyLSPTextDocumentContentChangeEvent(0xc0000d20c0, 0xc000170500, 0x1, 0x4, 0x4, 0xc000028701, 0x0)
    /home/nomad/Code/arduino-language-server/handler/textutils/textutils.go:13 +0x7c
github.com/arduino/arduino-language-server/handler.(*InoHandler).didChange(0xc0001cf7c0, 0xae9820, 0xc000024108, 0xc0001418c0, 0x2, 0x22, 0xc000504000)
    /home/nomad/Code/arduino-language-server/handler/handler.go:894 +0x12a
github.com/arduino/arduino-language-server/handler.(*InoHandler).HandleMessageFromIDE(0xc0001cf7c0, 0xae9820, 0xc000024108, 0xc0001de900, 0xc000101e50, 0x0, 0x0, 0x0, 0x0)
    /home/nomad/Code/arduino-language-server/handler/handler.go:343 +0x68d
github.com/sourcegraph/jsonrpc2.(*HandlerWithErrorConfigurer).Handle(0xc00020a250, 0xae9820, 0xc000024108, 0xc0001de900, 0xc000101e50)
    /home/nomad/go/pkg/mod/github.com/sourcegraph/jsonrpc2@v0.0.0-20200429184054-15c2290dcb37/handler_with_error.go:21 +0x72
github.com/sourcegraph/jsonrpc2.(*Conn).readMessages(0xc0001de900, 0xae9820, 0xc000024108)
    /home/nomad/go/pkg/mod/github.com/sourcegraph/jsonrpc2@v0.0.0-20200429184054-15c2290dcb37/jsonrpc2.go:522 +0x57e
created by github.com/sourcegraph/jsonrpc2.NewConn
    /home/nomad/go/pkg/mod/github.com/sourcegraph/jsonrpc2@v0.0.0-20200429184054-15c2290dcb37/jsonrpc2.go:334 +0x1d5
```

`textDocument/signatureHelp` works fine, however.  I have confirmed that the `clangd` Language Server is working on my system.

### Expected behavior

`textDocument/didChange` should not crash the Language Server.

### Environment

- Language Server version (commit hash): 436276b7a27511d8b0bf72189093563bb1449daa
- CLI version used (output of `arduino-cli version`): `arduino-cli alpha Version: 0.17.0-arch Commit: ab58758347e10f22e07154d520d0223612674cab`
- OS and platform: Arch Linux / Arduino Uno

### Additional context

My editor is the nightly build of Neovim (v0.5.0-893-ga1a4dd34e).

I have uploaded the server logs to this gist:
https://gist.github.com/michaeljoelphillips/85b51e740d11e0f9f7ce3f7bccf50add

rudism, 2022-06-18:
I'm running into this problem trying to use arduino-language-server as a neovim LSP provider as well. It works okay, providing diagnostics and autocompletes for a few minutes, but eventually segfaults.

<details>
<summary>Logs</summary>

```
[ERROR][2022-06-17 18:13:31] .../vim/lsp/rpc.lua:420	"rpc"	"arduino-language-server"	"stderr"	"2022/06/17 18:13:31 Panic: runtime error: invalid memory address or nil pointer dereference

goroutine 21 [running]:
runtime/debug.Stack()
	/usr/lib/go/src/runtime/debug/stack.go:24 +0x65
github.com/arduino/arduino-language-server/streams.CatchAndLogPanic()
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/streams/panics.go:14 +0x74
panic({0xa2cc20, 0xfdea30})
	/usr/lib/go/src/runtime/panic.go:838 +0x207
github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).TextDocumentDidChangeNotifFromIDE(0xc00016b6b0, {0xbecf20, 0xc00000e300}, 0xc000337290)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:1093 +0x8ae
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).TextDocumentDidChange(0xc00001f780?, {0xbecf20?, 0xc00000e300?}, 0x9d3840?)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/lsp_server_ide.go:256 +0x27
go.bug.st/lsp.(*Server).notificationDispatcher(0xc000151560, {0xbecf20, 0xc00000e300}, {0xc000398f90, 0x16}, {0xc00001f780, 0x71, 0x80})
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:208 +0x9a8
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingNotification(0xc000192b80, 0xc00042c580)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:183 +0x1e3
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingData(0xc000192b80, {0xc0005304d0, 0xaf, 0xaf})
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:121 +0x156
go.bug.st/lsp/jsonrpc.(*Connection).Run(0xc000192b80)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:110 +0x92
go.bug.st/lsp.(*Server).Run(...)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:116
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).Run(...)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/lsp_server_ide.go:34
github.com/arduino/arduino-language-server/ls.NewINOLanguageServer.func1()
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:148 +0x5b
created by github.com/arduino/arduino-language-server/ls.NewINOLanguageServer
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:146 +0x59c

"
[ERROR][2022-06-17 18:13:31] .../vim/lsp/rpc.lua:420	"rpc"	"arduino-language-server"	"stderr"	"panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x9804ce]

goroutine 21 [running]:
github.com/arduino/arduino-language-server/streams.CatchAndLogPanic()
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/streams/panics.go:15 +0x145
"
[ERROR][2022-06-17 18:13:31] .../vim/lsp/rpc.lua:420	"rpc"	"arduino-language-server"	"stderr"	"panic({0xa2cc20, 0xfdea30})
	/usr/lib/go/src/runtime/panic.go:838 +0x207
github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).TextDocumentDidChangeNotifFromIDE(0xc00016b6b0, {0xbecf20, 0xc00000e300}, 0xc000337290)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:1093 +0x8ae
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).TextDocumentDidChange(0xc00001f780?, {0xbecf20?, 0xc00000e300?}, 0x9d3840?)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/lsp_server_ide.go:256 +0x27
go.bug.st/lsp.(*Server).notificationDispatcher(0xc000151560, {0xbecf20, 0xc00000e300}, {0xc000398f90, 0x16}, {0xc00001f780, 0x71, 0x80})
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:208 +0x9a8
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingNotification(0xc000192b80, 0xc00042c580)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:183 +0x1e3
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingData(0xc000192b80, {0xc0005304d0, 0xaf, 0xaf})
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:121 +0x156
go.bug.st/lsp/jsonrpc.(*Connection).Run(0xc000192b80)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:110 +0x92
go.bug.st/lsp.(*Server).Run(...)
	/home/rudism/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:116
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).Run(...)
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/lsp_server_ide.go:34
github.com/arduino/arduino-language-server/ls.NewINOLanguageServer.func1()
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:148 +0x5b
created by github.com/arduino/arduino-language-server/ls.NewINOLanguageServer
	/home/rudism/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20211213092910-c1328634edf7/ls/ls.go:146 +0x59c
"
```

</details>

james2doyle, 2022-07-04:
Having the same issue in Sublime Text with Sublime LSP:

```
2022/07/04 12:22:50 [97mLS: : Initial board configuration: esp8266:esp8266[0m
2022/07/04 12:22:50 [97mLS: : Language server build path: /private/var/folders/n9/yf1zbxb945ngq8dzqvpnc4k80000gn/T/arduino-language-server686999688[0m
2022/07/04 12:22:50 [97mLS: : Language server build sketch root: /private/var/folders/n9/yf1zbxb945ngq8dzqvpnc4k80000gn/T/arduino-language-server686999688/sketch[0m
2022/07/04 12:22:50 [97mLS: : Language server compile-commands: /private/var/folders/n9/yf1zbxb945ngq8dzqvpnc4k80000gn/T/arduino-language-server282480269/compile_commands.json[0m
2022/07/04 12:22:50 [92mIDE --> LS REQU initialize 1[0m
2022/07/04 12:22:50 [96mINIT --- : initializing workbench: [0m
2022/07/04 12:22:50 Panic: runtime error: invalid memory address or nil pointer dereference

goroutine 20 [running]:
runtime/debug.Stack(0x47d76db, 0x2, 0xc000075950)
	/usr/local/go/src/runtime/debug/stack.go:24 +0x9f
github.com/arduino/arduino-language-server/streams.CatchAndLogPanic()
	/home/build/streams/panics.go:14 +0xc7
panic(0x46ec400, 0x4c96330)
	/usr/local/go/src/runtime/panic.go:971 +0x499
github.com/arduino/go-paths-helper.(*Path).Clone(...)
	/go/pkg/mod/github.com/arduino/go-paths-helper@v1.6.1/paths.go:81
github.com/arduino/go-paths-helper.(*Path).Canonical(0x0, 0x1)
	/go/pkg/mod/github.com/arduino/go-paths-helper@v1.6.1/paths.go:577 +0x2e
go.bug.st/lsp.DocumentURI.AsPath(0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/uri.go:36 +0xb4
github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).InitializeReqFromIDE.func1(0xc0000d5810, 0xc00013e000)
	/home/build/ls/ls.go:165 +0x1f8
created by github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).InitializeReqFromIDE
	/home/build/ls/ls.go:157 +0x65
```