arduino/arduino-language-server #80: How to set a project specific board-name

listout, 2021-09-11:
Looking at the `--help` command of the LS, it tells that we can change the `-board-name` variable using or the `-fqbn` variable using JSON-RPC. I'm not sure how to do this.

It would be extremely helpful if any maintainer/contributor could provide an example.

listout, 2021-09-11:
Also if I change the board-name or -fqbn manually, completion seems to stop working. I have tested this with the ESP32 boards, as soon as I change the `-board-name` or the `-fqbn` variable, completion stops working.

I'm on vim, using the coc.nvim pluging. My configuration for the server was;

```json
		"arduino": {
			"command": "~/.local/bin/arduino-language-server",
			"rootPatterns": [
				"*.ino"
			],
			"filetypes": [
				"arduino"
			],
			"args": [
				"-cli",
				"arduino-cli",
				"-cli-config",
				"~/.arduino15/arduino-cli.yaml",
				"-clangd",
				"clangd",
				"-board-name",
				"ESP32 Dev Module",
				"-fqbn",
				"esp32:esp32:esp32"
			]
		}
```

Woocash9, 2021-09-28:
I'm in the same situation, though I'm not using coc but nvim native lsp implementation. 
When I set manually -fqbn in the setup, completion works without any problem (first completion is quite slow as lsp-server is compiling sketch with arduino-cli to produce compilation-database, after that all works fine). 
Obviously it wood be far better to be able to change the board interactively. I have spent last 2 hours analyzing the code and it looks like there is no such possibility at the moment (despite what is stated in the --help message). 
I will try to implement this functionality tomorrow. Will let you know if it works.

RadekRojik, 2022-01-20:
> 

The same problem

S4deghN, 2022-02-12:
Running `arduino-cli board attach [fqbn]` in the root directory auto generated `sketch.json` which works and simply contains:
```
{
  "cpu": {
    "fqbn": "[fqbn]",
    "port": ""
  }
}
```
I'm using nvim-lsp and I've installed the lsp via LspInstaller, don't know if that makes a difference.

bradcush, 2022-06-11:
> I'm using nvim-lsp and I've installed the lsp via LspInstaller, don't know if that makes a difference.

I'm using Neovim LSP and have installed arduino-language-server using `go install` as well as built it directly from source and can confirm that the presense of the `sketch.json` allows you to ignore this flag when starting the language server. There is nothing special regarding setup using LspInstaller that is required to make this work. Personally I'm also not specifying the `-board-name` flag anywhere either and all language server features are working.

Taken from the project `README.md`, omitting the `-fqbn` flag:
``` sh
./arduino-language-server \
 -clangd /usr/local/bin/clangd \
 -cli /usr/local/bin/arduino-cli \
 -cli-config $HOME/.arduino15/arduino-cli.yaml
```

alessio-perugini, 2023-08-05:
:wave: We dropped support for sketch.json in favour of sketch.yaml.
If in your path you have the sketch.yaml and you have defined `default_fqbn: your_fqbn` it will be automatically picked up, without needing to specific the `-fqbn` flag in the `arduino-language-server`.

For the one using neovim that wants to do some magic you can script a bit like:
```lua
    on_new_config = function(config, root_dir)
        local file = io.open(root_dir .. '/sketch.yaml', "r")
        if file == nil then
            return
        end

        local fqbn = "arduino:avr:uno"
        local lines = file:lines()
        for line in lines do
            if string.find(line, "fqbn") then
                fqbn = string.gsub(line, "%s+fqbn:%s", "")
            end
        end

        file:close()

        config.cmd = {
            "arduino-language-server",
            "-fqbn", fqbn,
        }
    end,
    cmd = {
        "arduino-language-server",
        "-fqbn", "arduino:avr:uno",
    }
``` 

<details><summary>sketch.yaml</summary>
<p>

```yaml
profiles:
  portenta_c33:
    fqbn: arduino:renesas_portenta:portenta_c33
    platforms:
      - platform: arduino:renesas_portenta (1.0.2)
```

</p>
</details>