arduino/arduino-language-server #84: Nothing happens after I run this command ./arduino-language-server -cli-config <path-to-cli-config>

PaoGon, 2021-09-28:
## Bug Report
Nothings happens when I set the path to ArduinoCLI config

### Current behavior

go get -u github.com/arduino/arduino-language-server
go build
./arduino-language-server -cli-config $HOME/.arduino15/arduino-cli.yaml (nothings happen here)

I didnt get any response after I run this command ./arduino-language-server -cli-config $HOME/.arduino15/arduino-cli.yaml
so I just always hit ^C then the terminal always says this "2021/09/28 17:12:17 INTERRUPTED
2021/09/28 17:12:17 removing buildpath"

### Expected behavior
I dont really know what must happen after I run ./arduino-language-server -cli-config <path-to-cli-config> because there is no 
example on the documentation

### Environment

- Language Server version (commit hash): bbba79f98f1c3dcfe9c06e29095b3aff9de8f33d
- CLI version used (output of `arduino-cli version`):arduino-cli alpha Version: 0.19.0 Commit: 56419ecd Date: 2021-09-02T14:47:35Z
- OS and platform: Ubuntu 20.04.2

dylanashcraft, 2021-12-07:
Issue:
M1 Architecture, same results as PaoGon. arduino-language-server -cli-config does not accept relative or full path as string. No results after waiting 5 minutes.

**Input:**
`./arduino-language-server -cli-config ~/Library/Arduino15/arduino-cli.yaml`
**Output:**
```
^C2021/12/07 04:43:06 INTERRUPTED
2021/12/07 04:43:06 removing buildpath
```

husseljoo, 2021-12-10:
I have the same issue. I tried left it for as long as 4hrs and still no response.

husseljoo, 2021-12-11:
@per1234 is there any workaround to this?

cmaglie, 2021-12-11:
It's normal that you don't see any output because the language server is waiting for the `initialize` command. Anyway, it's not intended for usage by humans via the command line as you expect.

The Arduino language server is an implementation of the LSP (Language Server Protocol) specification: https://microsoft.github.io/language-server-protocol/specifications/specification-current/. The purpose of this program is to provide language-related functionality to the IDE's so, in general, it's the IDE that talks to the language server via stdin/stdout using the slightly modified JSONRPC protocol defined in the LSP specification.

Hope this helps, we should probably explain this better in the README, so I'll keep this issue open until then.

husseljoo, 2021-12-11:
Is there any guide on how to make the arduino-language-server to work on nvim lsp? It seems that cli-config <file-path> has to be included/declared somehow. How does one go about this?

cmaglie, 2021-12-11:
> Is there any guide on how to make the arduino-language-server to work on nvim lsp? It seems that cli-config has to be included/declared somehow. How does one go about this?

Nope, sorry, we are developing mainly for the Arduino IDE, I've no idea how to configure nvim...

Anyway, if you want to give it a try yourself, you need to put together three pieces of the puzzle:

- install arduino-cli (download the latest version and install the cores for your board via `arduino-cli core install ...`)
- install clangd (we use the build here https://github.com/clangd/clangd/releases/tag/13.0.0)
- make a configuration of nvim to let it start arduino-language-server correctly (here I can't help... you are on your own)

The Arduino language server uses clangd under the hood, that's the reason why it is required.

I suggest to build the arduino-language-server from this branch: https://github.com/arduino/arduino-language-server/pull/89 that has a TON of bugfixes

On the Arduino IDE the command line we use to start the language server is something like:

```
./arduino-language-server \
 -clangd /usr/local/bin/clangd \     <--- path to clang
 -cli /usr/local/bin/arduino-cli \      <--- path to arduino-cli
 -cli-config $HOME/.arduino15/arduino-cli.yaml \      <---- arduino-cli config
 -fqbn arduino:mbed:nanorp2040connect          <---- board FQBN
```

you must configure nvim to do something similar when starting the language server.
The `-fqbn` flag represents the board you're actually working on (different boards may implement different features/API, if you change board you need to restart the language server with another fqbn).

If everything goes right you should be able to edit .ino files and ask for completions, get errors/warnings in real time, etc. etc.

BTW keep in mind that the language server is not 100% accurate, it's really tricky, and we are getting a lot of bug reports.

If you want to see some logging you can add the flags:
```
 -log -logpath $HOME/Logs/
```
this will make the language server output a lot of logging data in the specified folder.

PaoGon, 2021-12-24:
Sorry for my late response I already fixed this. So it seems that we do not need to run this on the terminal  `./arduino-language-server -cli-config $HOME/.arduino15/arduino-cli.yaml`,  we just need to put this on the arduino language server setup in neovim. 

require'lspconfig'.arduino_language_server.setup({ 
    cmd =  { 
        -- Required
        "arduino-language-server",
        "-cli-config", "$HOME/.arduino15/arduino-cli.yaml",
        "-cli", "/usr/bin/arduino-cli",
        "-clangd", "/usr/bin/clangd"
   }
})

per1234, 2022-06-22:
The readme has since been updated to explain this situation: https://github.com/arduino/arduino-language-server/commit/52d96f02aa3a70169e117d633fb92349cd41ae36

It will also be explained by the tool output itself: https://github.com/arduino/arduino-language-server/pull/119