arduino/arduino-language-server #97: Completion not working?

HendrikMennen, 2022-01-04:
## Bug Report

I am not sure if this happens with Arduino IDE too, but for me Completion does not work at all in the latest build.

It shows that completion is requested, but I never get a response.
Clangd alone works fine with my IDE. 
```
[00:03:10] Arduino LS: 2022/01/04 14:35:54 ←[92m                 textDocument/didChange: ←[93mwrite-unlocked←[0m←[0m
[00:03:10] Arduino LS: 2022/01/04 14:35:54 ←[92mIDE --> LS REQU textDocument/completion 3←[0m
[00:03:10] Arduino LS: 2022/01/04 14:35:54 ←[91mIDE     LS <-- Clangd NOTIF textDocument/publishDiagnostics←[0m
```

listout, 2022-02-03:
Having the same issue. The server is not sending any response.

drank40, 2022-03-13:
I have the same issue while using the language server with neovim. I think this is more than a mere imperfection and a pretty serious issue, especially if it also affects the official IDE. Its been over 2 months and i still get this error with the git version of the language server. See the issue mentioned above

HendrikMennen, 2022-09-07:
Any updates on this?

listout, 2024-04-28:
> almost two years later... nothing new?

I have not tested it yet but nvim-lspconfig seems to have a decent tutorial
on using the Arduino language sever. You can refer
https://github.com/neovim/nvim-lspconfig/blob/master/doc/server_configurations.md#arduino_language_server

metalinspired, 2024-04-29:
My set up is just fine since I have a dozen of other LSPs running :)
The issue is that Arduino Language Server simply does not reply while initializing.

Here is a error from LSP log:
`[ERROR][2024-04-29 07:14:22] .../vim/lsp/rpc.lua:770	"rpc"	"/home/foo/.arduino-language-server/arduino-language-server"	"stderr"	"07:14:21.817051 \27[97mLS: : Initial board configuration: \27[0m\n07:14:21.817088 \27[97mLS: : arduino-language-server Version: 0.0.0-git Commit:  Date: \27[0m\n07:14:21.817097 \27[97mLS: : Language server temp directory: /tmp/arduino-language-server1081938582\27[0m\n07:14:21.817099 \27[97mLS: : Language server build path: /tmp/arduino-language-server1081938582/build\27[0m\n07:14:21.817101 \27[97mLS: : Language server build sketch root: /tmp/arduino-language-server1081938582/build/sketch\27[0m\n07:14:21.817105 \27[97mLS: : Language server FULL build path: /tmp/arduino-language-server1081938582/fullbuild\27[0m\n07:14:21.818941 IN Elapsed: 1.775891ms\n07:14:21.819698 \27[92mIDE --> LS REQU initialize 1\27[0m\n"`

My Go skills are way too rusty otherwise I would try to fix it my self.