arduino/arduino-language-server #99: No support for neovim / Coc

beyarkay, 2022-03-13:
## Bug Report

### Current behavior

Currently, there's no way to get language server in neovim via Coc

### Expected behavior

It would be amazing to have hints and language support in neovim via Coc

### Environment

- Language Server version (commit hash): None applicable
- CLI version used (output of `arduino-cli version`): None applicable
- OS and platform: MacOS 12

drank40, 2022-03-13:
It used to work flawlessy, this is a regression

beyarkay, 2022-03-13:
@drank40 can you confirm the steps you used to you used to install it as a language server?

drank40, 2022-03-16:
I followed 
https://github.com/neoclide/coc.nvim/wiki/Language-servers#[Arduino](https://github.com/neoclide/coc.nvim/wiki/Language-servers#Arduino)

I just added this lines to my coc-settings.json and it worked, but it appears to be broken now, since the language server closes the connection at every request

drank40, 2022-03-16:
See 

https://github.com/neoclide/coc.nvim/issues/3594#issuecomment-1046250743

steprobe, 2022-05-09:
I am seeing the same thing on Ubuntu. Is this really an enhancement? Seems more like a bug, and as @drank40 points out a regression.

atrosekta, 2022-07-28:
same here too, coc's logs says the server is repeatedly stopping as soon as it start ,same with nvim lsp, tho i can launch the server "by hand" without it stopping

chmanie, 2022-07-28:
For me, using `nvim-lsp`, it does run for _some time_ but then crashes intermittently. Are we talking about the same issue or is this different behavior?

dbmzzo, 2022-08-06:
> For me, using `nvim-lsp`, it does run for _some time_ but then crashes intermittently.

This is the behavior I'm seeing as well, using `nvim-lspconfig`.

beyarkay, 2022-08-06:
@per1234 @cmaglie is this issue likely to be fixed in the near future? Or alternatively, is the repository accepting PRs?

per1234, 2022-08-06:
> is the repository accepting PRs?

Yes. We have already received some high quality PRs which resulted in valuable contributions to the Arduino Language Server project.

One thing that should be noted is that the primary purpose of this project is to serve as the language server component of [Arduino IDE 2.x](https://github.com/arduino/arduino-ide). The use of the standardized language server protocol makes it a general purpose tool that can be used in other applications and it is very cool that people are already using it in such alternative applications. However, that use case will always be a lower priority for the project relative to the use in the development tools created by Arduino such as Arduino IDE 2.x.

beyarkay, 2022-08-06:
So reading between the lines, support for non-arduino editors is not available, but the project is willing to accept (high quality) PRs if they add that support?

(I'll take silence to mean a yes)

beyarkay, 2022-08-06:
Sorry I'm realising that comment could come across a little aggressive, which is not my intention at all. Just wanting to get a clear idea of the state of support for non-arduino features before I go and try fix the problem myself.

Thanks for the help!

adamansky, 2022-08-20:
All what you need is a proper `compile_commands.json` in your sketch directory and CoC + cland installed with vim/neovim

Take look on my small utility project which brings me code completion for Arduino projects 
https://github.com/Softmotions/acdb

attilastrba, 2022-09-02:
> All what you need is a proper `compile_commands.json` in your sketch directory and CoC + cland installed with vim/neovim

Can you explain a bit more? I have a proper `compile_commands.json` still the server is crashing. I am using Neovim, `arduino_cli v0.26.0` and compiled the language server from scratch. If I don't provide the arduino_cli path then is all ok. I am unable to get any log with the -log function. 

Thank you

adamansky, 2022-09-03:
> still the server is crashing

Along with `compler_commans.json` in project root just use [coc-clangd](https://github.com/clangd/coc-clangd) as a language server and associate `.ini` files with `cpp`

```vim
let g:coc_filetype_map = {
  \ 'ino': 'cpp',
  \ }
```

attilastrba, 2022-09-05:
Thank you. You mean that I don't need the whole `arduino-language-server` stuff and just work with clangd?

adamansky, 2022-09-05:
Sure!

per1234, 2023-06-26:
Closing a duplicate of the other issues we have about neovim/coc:

- https://github.com/arduino/arduino-language-server/issues/155
- https://github.com/arduino/arduino-language-server/issues/159
- https://github.com/arduino/arduino-language-server/issues/152

closed 2023-06-26 by per1234