arduino/arduino-language-server #100: Feature Request: read sketch.json for project-specific settings (e.g. FQBN)

nalzok, 2022-03-20:
Requiring a `-fqbn` option on the command line makes it difficult to set project-specific board names, and [the current workaround](https://github.com/williamboman/nvim-lsp-installer/blob/main/lua/nvim-lsp-installer/servers/arduino_language_server/README.md#dynamically-changing-fqbn-per-project) calls for hardcoded FQBN for each individual project, which is inconvenient and error-prone. It would be great if arduino-language-server could accept a `-project-root` option and check the `sketch.json` inside the project root directory to access all project-specific settings.

bradcush, 2022-06-12:
This should already be the case if your `sketch.json` file is at the same level as your `ino` file. In my case I'm using `arduino_language_server` with the Neovim built-in LSP which sets the root directory for the language server based on the location of the `ino` file. I haven't dug too deep but I have one language server client per root directory. I'm guessing if the language server process is started from a directory that contains a `sketch.json` file then it will use it.

alessio-perugini, 2023-08-05:
@nalzok it is possible. Just create a `sketch.yaml` in the root of your project and add the following key: `default_fqbn: your_fqbn`. By doing so you can change your neovim lspconfig removing the custom function and the `-fqbn` flag.

```yaml
default_fqbn: arduino:avr:uno
```