arduino/arduino-language-server #109: Server Fails with Neovim LSP - `ArduinoCLI daemon address and instance number must be set.`

nsapse, 2022-06-04:
### Describe the problem

When the server is installed according to the instructions [given here in the lspconfig documentation](https://github.com/neovim/nvim-lspconfig/blob/master/doc/server_configurations.md#arduino_language_server=) the server fails with the error `[ERROR][2022-06-04 15:32:37] .../vim/lsp/rpc.lua:420    "rpc"   "arduino-language-server"   "stderr"    "2022/06/04 15:32:37 ArduinoCLI daemon        address and instance number must be set.\n"` (found in LSP logs). This is not fixed by adding the additional requested flags for daemon and instance number (i.e. `"-cli-daemon-addr", "localhost:5051",    
"-cli-daemon-instance", "1",   `) does not address this problem.          

### To reproduce

1. Install neovim 0.7.0 and install lspconfig.
2. Add arduino_language_server per the instructions posted above.
3. Create a `test.ino` file or open an existing `.ino` file with Neovim.

### Expected behavior

Server should launch and attach.

### Arduino Language Server version

0.6.0

### Arduino CLI version

arduino-cli  Version: 0.22.0 Commit: 65f662a7 Date: 2022-05-06T11:09:35Z

### Operating system

Linux

### Operating system version

Latest Manjaro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

bradcush, 2022-06-11:
The error `ArduinoCLI daemon address and instance number must be set.` is a bit misleading, you shouldn't need to set those flags. I found this error to happen when you are missing other required flags. In addition to `-cli-config` you need to specify the path to the `clangd` binary using`-clangd` and the path to the `arduino-cli` binary using `-cli`.

I'm not sure what the full command you're running is to start the language server but you should be able to test it yourself using the example directly from the project README.md, just replacing the paths where you need to. I find the `nvim-lspconfig` docs to be misleading, they mention some flags as optional which don't actually seem to be.

``` sh
./arduino-language-server \
 -clangd /usr/local/bin/clangd \
 -cli /usr/local/bin/arduino-cli \
 -cli-config $HOME/.arduino15/arduino-cli.yaml \
 -fqbn arduino:mbed:nanorp2040connect
```

nsapse, 2022-06-11:
What is the expected output of that example command?

bradcush, 2022-06-11:
> What is the expected output of that example command?

That command should start the language server running in a foreground process:

``` text
17:23:45.616289 LS: : Initial board configuration:
17:23:45.616419 LS: : Language server build path: /private/var/folders/cl/h4959ym96wn_dd1zg6v7tdl00000gq/T/arduino-language-server1806086785
17:23:45.616427 LS: : Language server build sketch root: /private/var/folders/cl/h4959ym96wn_dd1zg6v7tdl00000gq/T/arduino-language-server1806086785/sketch
17:23:45.616431 LS: : Language server compile-commands: /private/var/folders/cl/h4959ym96wn_dd1zg6v7tdl00000gq/T/arduino-language-server1401792245/compile_commands.json
```

nsapse, 2022-06-11:
A few interesting things about that:
1) The LSP-Config docs are definitely misleading based on your comment. The Arduino LSP team should probably open an issue (or I'd be happy to) regarding that.

2) The provided command works whether or not  the `clangd` and `arduino-cli` binaries are being referenced in the correct locations (e.g. my clangd is in  in `/usr/bin/clangd` not `/usr/local/bin/clangd` but feeding either into the flag results in the same input you posted above).

3) Despite the example working properly when launching the server as a foreground process the server still fails to attach to NVIM when using the same command, although now  with a totally different error (see below):

(`[ERROR][2022-06-11 18:11:57] .../vim/lsp/rpc.lua:420    "rpc"   "arduino-language-server"   "stderr"    "2022/06/11 18:11:57 \27[92mIDE --> LS NOTIF initialized\27[0m\n2022/06/11 18:11:57 \27[92m                 initialized: Notification is not propagated to clangd\27[0m\n2022/06/11 18:11:57 \27[92mIDE        --> LS NOTIF workspace/didChangeConfiguration\27[0m\n2022/06/11 18:11:57 Panic: unimplemented\n\ngoroutine 8 [running]:\nruntime/debug.Stack(0xbd723a, 0x2, 0xc0002bdac8)\n\t/usr/local/go/src/runtime/debug/stack.go:24 +0x9f\ngithub.com/arduino/arduino-language-server/streams.CatchAndLogPanic()\n\t/home       /build/streams/panics.go:14 +0xc7\npanic(0xaaf920, 0xc98920)\n\t/usr/local/go/src/runtime/panic.go:965 +0x1b9\ngithub.com/arduino/arduino-language-server/ls.(*IDELSPServer).WorkspaceDidChangeConfiguration(0xc0001e51e0, 0xca5fe0, 0xc00013c048, 0xc00013c060)\n\t/home/build/ls/lsp_server_ide.go:232 +0x39\       ngo.bug.st/lsp.(*Server).notificationDispatcher(0xc00004d560, 0xca5fe0, 0xc00013c048, 0xc0001340e0, 0x20, 0xc0001420c0, 0x2c, 0x30)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:166 +0xfde\ngo.bug.st/lsp/jsonrpc.(*Connection).handleIncomingNotification(0xc0000f2b80, 0xc00012       c040)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:183 +0x137\ngo.bug.st/lsp/jsonrpc.(*Connection).handleIncomingData(0xc0000f2b80, 0xc000146080, 0x74, 0x74)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:12       1 +0x245\ngo.bug.st/lsp/jsonrpc.(*Connection).Run(0xc0000f2b80)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:110 +0x97\ngo.bug.st/lsp.(*Server).Run(...)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:116\ngithub.com/arduino/ard       uino-language-server/ls.(*IDELSPServer).Run(...)\n\t/home/build/ls/lsp_server_ide.go:34\ngithub.com/arduino/arduino-language-server/ls.NewINOLanguageServer.func1(0xc0000cd970, 0xc00000eaf8)\n\t/home/build/ls/ls.go:148 +0x5b\ncreated by github.com/arduino/arduino-language-server/ls.NewINOLanguageServer\       n\t/home/build/ls/ls.go:146 +0x59b\n\npanic: unimplemented [recovered]\n\tpanic: unimplemented\n\ngoroutine 8 [running]:\ngithub.com/arduino/arduino-language-server/streams.CatchAndLogPanic()\n\t/home/build/streams/panics.go:15 +0x225\npanic(0xaaf920, 0xc98920)\n\t/usr/local/go/src/runtime/panic.go:965        +0x1b9\ngithub.com/arduino/arduino-language-server/ls.(*IDELSPServer).WorkspaceDidChangeConfiguration(0xc0001e51e0, 0xca5fe0, 0xc00013c048, 0xc00013c060)\n\t/home/build/ls/lsp_server_ide.go:232 +0x39\ngo.bug.st/lsp.(*Server).notificationDispatcher(0xc00004d560, 0xca5fe0, 0xc00013c048, 0xc0001340e0, 0x       20, 0xc0001420c0, 0x2c, 0x30)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:166 +0xfde\ngo.bug.st/lsp/jsonrpc.(*Connection).handleIncomingNotification(0xc0000f2b80, 0xc00012c040)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:18       3 +0x137\ngo.bug.st/lsp/jsonrpc.(*Connection).handleIncomingData(0xc0000f2b80, 0xc000146080, 0x74, 0x74)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:121 +0x245\ngo.bug.st/lsp/jsonrpc.(*Connection).Run(0xc0000f2b80)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20       211202163946-3ad3994172a0/jsonrpc/jsonrpc_connection.go:110 +0x97\ngo.bug.st/lsp.(*Server).Run(...)\n\t/go/pkg/mod/go.bug.st/lsp@v0.0.0-20211202163946-3ad3994172a0/server.go:116\ngithub.com/arduino/arduino-language-server/ls.(*IDELSPServer).Run(...)\n\t/home/build/ls/lsp_server_ide.go:34\ngithub.com/ar       duino/arduino-language-server/ls.NewINOLanguageServer.func1(0xc0000cd970, 0xc00000eaf8)\n\t/home/build/ls/ls.go:148 +0x5b\ncreated by github.com/arduino/arduino-language-server/ls.NewINOLanguageServer\n\t/home/build/ls/ls.go:146 +0x59b\n"
~`

bradcush, 2022-06-11:
Is this the only error that's output, I would imagine there are others as well. I had multiple errors when I was able to get the language server to properly attach to the buffer but it still wasn't working. When you check `:LspInfo`, you don't have the `arduino_language_server` showing as attached to the buffer? I would expect if the command tested in the shell starts the language server and it continues to run without interruption it should at least be attached. There just might be additional issues preventing it from working properly, one of which you've listed above.

Here's a simplified example for reference of what I have in my LSP configuration. To simplify things and hopefully get you to a point where it's running, I've specified the `-fqbn` flag hardcoded to `arduino:avr:uno`. This should be project specific but I'm listing it here as it should remove some variables when trying to determine what's wrong.

``` lua
-- Set paths for arduino language server command
local arduino_binary = '/Users/bcushing/go/bin/arduino-language-server'
local arduino_config = '/Users/bcushing/Library/Arduino15/arduino-cli.yaml'
nvim_lsp['arduino_language_server'].setup({
    on_attach = on_attach,
    cmd = {
        arduino_binary,
        '-cli-config',
        arduino_config,
        '-clangd',
        'clangd',
        '-cli',
        'arduino-cli',
        '-fqbn',
        'arduino:avr:uno'
    }
})
```

nsapse, 2022-06-12:
After playing with this for a while longer I finally got it to attach. The issue appears to be that there was some sort of negative interaction between the server and [nvim-lsp-installer](https://github.com/williamboman/nvim-lsp-installer). Separate from my LSP logs I found issues where nvim-lsp-installer was labeling the startup command as non-executable for some reason despite my never having tried to use to to set up Arduino LSP. Removing the plugin and using manual setup with the config you posted works flawlessly. Will close this issue. Thank you for all the help.

closed 2022-06-12 by nsapse