arduino/arduino-language-server #127: Add CI workflows to enhance the dev experience

umbynos, 2022-07-29:
Add some checks workflows to this repo to enhance the developer experience and to catch common problems:
- https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-go-task.md
- https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-go-dependencies-task.md
- https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-license.md
- https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-markdown-task.md
- https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-taskfiles.md

per1234, 2022-07-29:
@umbynos @MatteoPologruto. I have already prepared the work for https://github.com/arduino/tooling-project-assets/blob/main/workflow-templates/check-go-dependencies-task.md last year, but have been blocked on its completion waiting for an answer to a question. See `#7218` in the internal issue tracker for details.

The other workflows will be super valuable additions to the project though.

MatteoPologruto, 2022-08-04:
Check workflows additions:
- Check-go-task: PR #128 
- Check-taskfiles: PR #129 
- Check-markdown: PR #131 
- Check-license: PR #132 

Minor updates:
- Windows runner updated to `windows-2019`: PR #130

per1234, 2022-08-04:
Thanks so much for your excellent work Matteo.

Since we already have an issue in the internal tracker for adding the remaining workflow, I will close this as resolved.

closed 2022-08-04 by per1234