arduino/arduino-language-server #144: Random memory leak when typing Chinese comments in my sketch

Silver-Fang, 2022-12-24:
### Describe the problem

When I type Chinese comments in my sketch, severe memory leak occurs randomly in the arduino-language-server process.
[image]

### To reproduce

Since the issue occurs randomly, there are no steps to 100% reproduce. But after some fumbling, I found a way to reproduce with the greatest probability:
1. Open my sketch with Arduino IDE v2.0.3, and switch to IStepTrial.h
2. Open Task Manager and monitor the memory footprint of arduino-language-server.
3. Use the Microsoft Pinyin input method to type the comment text on line 16 as it is on line 17. Note that keyboard input is required, not copy-paste. You will most likely observe a memory explosion before you finish this line.
[image]

[Gbec.zip](https://github.com/arduino/arduino-language-server/files/10297932/Gbec.zip)

I have tried some of other sketches and they were all less likely to trigger this issue than my sketch. I offer my sketch here so that you may easily reproduce the issue.

### Expected behavior

The Arduino code language may not have plans to support Chinese, but I don't think there should be a serious problem like a memory leak just because of a few lines of Chinese comments anyway.

### Arduino Language Server version

0.7.2

### Arduino CLI version

0.29.0

### Operating system

Windows

### Operating system version

Win11 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

Silver-Fang, 2023-09-07:
This issue has been extremely rare in recent releases. While not sure where the update happened to fix this issue as a side effect?

14725, 2024-02-05:
It is rare but still happens. However it seemed to triggered the OOM Killer of my Ubuntu...

Keeping track on this issue... 

------

```
版本：2.2.1
日期：2023-08-31T14:16:33.536Z
CLI 版本：0.34.0

Copyright © 2024 Arduino SA
```

```
Arduino AVR Boards 1.8.6
```

I'm using the AppImage release of Arduino.

14725, 2024-02-05:
It can be reproduced easily with the Arduino version.
```
  //此方法不是必须实现。如果你的步骤是即时步骤，无需实现此方法，因为即时步骤立即完成，不会被暂停。但如果是延时步骤，建议实现此方法以支持暂停。
```
->
```
//啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊啊此方法不是必须实现。如果你的步骤是即时步骤，无需实现此方法，因为即时步骤立即完成，不会被暂停。但如果是延时步骤，建议实现此方法以支持暂停。
```
\~8MB -> \~300MB -> \~79.2MB

[core_arduino_lang_serv.zip](https://github.com/arduino/arduino-language-server/files/14162991/core_arduino_lang_serv.zip)

Core dumps, maybe helpful to some hackers. (Warning: ZIP Bomb! ),