arduino/arduino-language-server #152: [nvim + arduino-language-server]: Clang-related error

oliverswitzer, 2023-03-07:
### Describe the problem

Hi,

I am seeing some odd logs that I can't quite make sense of that seem related to communication happening between arduino-language-server and my local clang install:

I am using nvim + nvim-config + lsp-installer to try and get arduino-language-server working for my local nvim setup.

When opening a file titled `fibonacci.ino`, I see the following output in the logs (I have turned on LSP tracing). There is additional output, but didn't want to post a wall of text, but let me know if you'd like to see more:

```
...
[DEBUG][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:365    "rpc.send"      {  id = 1,  jsonrpc = "2.0",  method = "initialize",  params = {    capabilities = {      callHierarchy = {        dynamicRegistration = false      },      textDocument = {        codeAction = {          codeActionLiteralSupport = {            codeActionKind = {              valueSet = { "", "Empty", "QuickFix", "Refactor", "RefactorExtract", "RefactorInline", "RefactorRewrite", "Source", "SourceOrganizeImports", "quickfix", "refactor", "refactor.extract", "refactor.inline", "refactor.rewrite", "source", "source.organizeImports" }            }          },          dataSupport = true,          dynamicRegistration = false,          isPreferredSupport = true,          resolveSupport = {            properties = { "edit" }          }        },        completion = {          completionItem = {            commitCharactersSupport = false,            deprecatedSupport = false,            documentationFormat = { "markdown", "plaintext" },            preselectSupport = false,            snippetSupport = false          },          completionItemKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25 }          },          contextSupport = false,          dynamicRegistration = false        },        declaration = {          linkSupport = true        },        definition = {          linkSupport = true        },        documentHighlight = {          dynamicRegistration = false        },        documentSymbol = {          dynamicRegistration = false,          hierarchicalDocumentSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        hover = {          contentFormat = { "markdown", "plaintext" },          dynamicRegistration = false        },        implementation = {          linkSupport = true        },        publishDiagnostics = {          relatedInformation = true,          tagSupport = {            valueSet = { 1, 2 }          }        },        references = {          dynamicRegistration = false        },        rename = {          dynamicRegistration = false,          prepareSupport = true        },        signatureHelp = {          dynamicRegistration = false,          signatureInformation = {            activeParameterSupport = true,            documentationFormat = { "markdown", "plaintext" },            parameterInformation = {              labelOffsetSupport = true            }          }        },        synchronization = {          didSave = true,          dynamicRegistration = false,          willSave = false,          willSaveWaitUntil = false        },        typeDefinition = {          linkSupport = true        }      },      window = {        showDocument = {          support = false        },        showMessage = {          messageActionItem = {            additionalPropertiesSupport = false          }        },        workDoneProgress = true      },      workspace = {        applyEdit = true,        configuration = true,        symbol = {          dynamicRegistration = false,          hierarchicalWorkspaceSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        workspaceEdit = {          resourceOperations = { "rename", "create", "delete" }        },        workspaceFolders = true      }    },    clientInfo = {      name = "Neovim",      version = "0.8.0"    },    initializationOptions = vim.empty_dict(),    processId = 74770,    rootPath = "/Users/oliverswitzer/workspace/arduino-playground",    rootUri = "file:///Users/oliverswitzer/workspace/arduino-playground",    trace = "off",    workspaceFolders = { {        name = "/Users/oliverswitzer/workspace/arduino-playground",        uri = "file:///Users/oliverswitzer/workspace/arduino-playground"      } }  }}
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.490985 clangd found at /usr/bin/clangd\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.491683 \27[97mLS: : Initial board configuration: arduino:avr:nano\27[0m\n20:59:50.491687 \27[97mLS: : arduino-language-server Version: 0.0.0-git Commit:  Date: \27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.491694 \27[97mLS: : Language server build path: /private/var/folders/km/y_f3tv3s0yz6f1h4z09qmr300000gn/T/arduino-language-server486595322\27[0m\n20:59:50.491699 \27[97mLS: : Language server build sketch root: /private/var/folders/km/y_f3tv3s0yz6f1h4z09qmr300000gn/T/arduino-language-server486595322/sketch\27[0m\n20:59:50.491701 \27[97mLS: : Language server FULL build path: /private/var/folders/km/y_f3tv3s0yz6f1h4z09qmr300000gn/T/arduino-language-server207618810\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.492205 IN Elapsed: 15.708µs\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.492540 \27[92mIDE --> LS REQU initialize 1\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.493056 \27[92m                initialize 1: \27[93mwrite-locked\27[0m\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.493147 \27[92m                initialize 1: \27[93mwrite-unlocked\27[0m\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.493188 \27[96mINIT --- : initializing workbench: file:///Users/oliverswitzer/workspace/arduino-playground\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.493401 \27[92m                initialize 1: initialization parameters: {\"capabilities\":{\"textDocumentSync\":{\"openClose\":true,\"change\":2,\"save\":{\"includeText\":true}},\"completionProvider\":{\"triggerCharacters\":[\".\",\"\\u003c\",\"\\u003e\",\":\",\"\\\"\",\"/\"],\"allCommitCharacters\":[\" \",\"\\t\",\"(\",\")\",\"[\",\"]\",\"{\",\"}\",\"\\u003c\",\"\\u003e\",\":\",\";\",\",\",\"+\",\"-\",\"/\",\"*\",\"%\",\"^\",\"\\u0026\",\"#\",\"?\",\".\",\"=\",\"\\\"\",\"'\",\"|\"],\"completionItem\":{}},\"hoverProvider\":{},\"signatureHelpProvider\":{\"triggerCharacters\":[\"(\",\",\"]},\"definitionProvider\":{},\"documentHighlightProvider\":{},\"documentSymbolProvider\":{},\"codeActionProvider\":{\"codeActionKinds\":[\"quickfix\",\"refactor\",\"info\"]},\"documentFormattingProvider\":{},\"documentRangeFormattingProvider\":{},\"documentOnTypeFormattingProvider\":{\"firstTriggerCharacter\":\"\\n\"},\"renameProvider\":{},\"executeCommandProvider\":{\"commands\":[\"clangd.applyFix\",\"clangd.applyTweak\"]},\"workspaceSymbolProvider\":{}},\"serverInfo\":{\"name\":\"arduino-language-server\",\"version\":\"0.0.0-git\"}}\27[0m\n"
[ERROR][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:440    "rpc"   "arduino-language-server"       "stderr"        "20:59:50.493409 \27[32mIDE <-- LS RESP initialize 1\27[0m\n20:59:50.493436 \27[96mINIT --- : \27[93mread-locked\27[0m\27[0m\n20:59:50.493440 \27[96mINIT --- : \27[93mread-unlocked\27[0m\27[0m\n20:59:50.493437 OUT Elapsed: 5.458µs\n"
[DEBUG][2023-03-06 20:59:50] .../vim/lsp/rpc.lua:474    "rpc.receive"   {  id = 1,  jsonrpc = "2.0",  result = {    capabilities = {      codeActionProvider = {        codeActionKinds = { "quickfix", "refactor", "info" }      },      completionProvider = {        allCommitCha [... cut ...]
`client 8 quit with exit code 2 and signal 0` in the nvim editor, and the log lines listed above in my `lsp.log` file.

### Arduino Language Server version

0.7.4

### Arduino CLI version

0.31.0 Commit: 940c9457 Date: 2023-02-20T15:09:19Z

### Operating system

macOS

### Operating system version

13.2.1 (22D68) - Ventura

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

saippua, 2023-03-23:
also experiencing this. Any workarounds?

WearsomeKarma, 2023-05-09:
Also experiencing this issue.

BrunoCooper17, 2023-06-02:
Try disabling lsp semantic tokens (https://www.reddit.com/r/neovim/comments/zjqquc/how_do_i_turn_off_semantic_tokens/). Apparently the server does not support them and are disabled by default (https://github.com/neovim/nvim-lspconfig/blob/master/doc/server_configurations.md#arduino_language_server, see capabilities)

kons-9, 2023-06-02:
I faced the same error, and I found this error is because of the way of finding a file name in arduino-cli.
The algorithm is "root directory" + "ino". For example, since your directory is arduino-playground, the language server tries to find arduino-playground.ino, not fibonacci.ino, which is the reason for the error.

per1234, 2023-06-26:
Thanks for explaining the cause of the reported error @kons-9. Arduino sketches must contain a `.ino` file with name matching the sketch folder:

https://arduino.github.io/arduino-cli/dev/sketch-specification/#primary-sketch-file

closed 2023-06-26 by per1234