arduino/arduino-language-server #160: arduino-language-server Version: 0.0.0

ubaldot, 2023-06-11:
### Describe the problem

When I run the LS through `./arduino-language-server -cli-config /Users/ubaldot/Library/Arduino15/arduino-cli.yaml -cli /opt/homebrew/bin/arduino-cli -fqbn arduino:avr:uno -clangd /usr/bin/clangd` from `zsh` shell I noticed a suspicious `arduino-language-server Version: 0.0.0` as shown in the following excerpt: 

$ ./arduino-language-server -cli-config /Users/ubaldot/Library/Arduino15/arduino-cli.yaml -cli /opt/homebrew/bin/arduino-cli -fqbn arduino:avr:uno -clangd /usr/bin/clangd
21:25:26.647772 LS: : Initial board configuration: arduino:avr:uno
21:25:26.647823 LS: : arduino-language-server Version: 0.0.0-git Commit:  Date:

I am suspicious that this cause the language client to` exit with status 2`. I am using the following client: https://github.com/yegappan/lsp 

### To reproduce

Execute the same steps as written the README file (download go, go build, etc). 
Use the following LS-client: https://github.com/yegappan/lsp and MacVim 9.0.1276, register the LS as: 

    {
        name: 'arduino-language-server',
        filetype: ['arduino'],
        path: $HOME .. '/Documents/arduino-language-server/arduino-language-server',
        args: ['-clangd /usr/bin/clangd',
                '-cli-config /Users/ubaldot/Library/Arduino15/arduino-cli.yaml',
                '-cli /opt/homebrew/bin/arduino-cli',
                '-fqbn arduino:avr:uno', '-log', '-logpath /Users/ubaldot/Documents']
    }

### Expected behavior

The LSP does not exit with status 2. 

### Arduino Language Server version

0.0.0 (should be 0.7.4 I guess)

### Arduino CLI version

arduino-cli  Version: 0.33.0 Commit: ca60d4b4 Date: 2023-05-29T15:18:18Z

### Operating system

macOS

### Operating system version

13.4 (22F66)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2023-06-11:
Hi @ubaldot. `0.0.0-git` is the fallback version number that is used when the version is not set via the build command:

https://github.com/arduino/arduino-language-server/blob/6c64232f29f8e612828b09767f0991f056e5c1e7/version/version.go#L21

If you want the version to be automatically populated, you should instead run this command from the root of your local clone of the arduino-language-server repository:

```text
task go:build
```

You will need to install the **Task** task runner tool dependency:

https://taskfile.dev/

But if you only want to use Arduino Language Server and don't have any interest in doing development work on the code base, then you should use the builds Arduino distributes.

If you want the release build, simply download the build from the "**Assets**" section of the latest release:

https://github.com/arduino/arduino-language-server/releases/latest

If you want to do beta testing, download the "**build-artifacts**" artifact from the latest scheduled run of the build workflow:

https://github.com/arduino/arduino-language-server/actions/workflows/build.yml?query=event%3Aschedule

> I am suspicious that this cause the language client to exit with status 2.

It is true that the build procedure specified in the readme does not follow the best practices for the project development. However, the only difference between the result of `go build` vs. `task go:build` is the automatic version number setting. That is purely a convenience feature in the context of a build that is used only for the purposes of local usage during development. There is no technical benefit of it. The tester and release builds of Arduino Language Server distributed by Arduino have the appropriate version number set.

So the fallback version number that results from building Arduino Language Server via `go build` will definitely not cause any problems for Arduino Language Server itself. I can't comment on whether a dependent application might run a version check.

---

Even though the current build instructions are serviceable. I do think it would make sense to add a comprehensive Contributor Guide which documents:

- Beta testing
- Development dependencies
- Ideal build, test, and validation procedures

This can be based on our standard Contributor Guide template:

https://github.com/arduino/tooling-project-assets/tree/main/documentation-templates#contributor-guide

ubaldot, 2023-06-12:
Thanks for the reply. i followed your instructions and I get the correct version number, however I still have the `exit with status 2` from the lsp client that I mentioned. Any hints on how to fix it?

closed 2025-03-19 by alessio-perugini