arduino/arduino-language-server #161: How to use it with Arduino IDE 2? README would need an update.

ubaldot, 2023-06-11:
### Describe the problem

In the README it is stated that the LSP can be used with Arduino IDE 2 but it is not written how. 
Could you please update the README?

### To reproduce

-

### Expected behavior

-

### Arduino Language Server version

-

### Arduino CLI version

-

### Operating system

N/A

### Operating system version

-

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2023-06-11:
Hi @ubaldot. Arduino Language Server is integrated into Arduino IDE 2.x, so there is nothing to document in the context of this repository. Just install the IDE and Arduino Language Server is installed as part of the Arduino IDE 2.x installation and automatically providing code-aware features for all the sketches you open in Arduino IDE 2.x.

You can learn how to install Arduino IDE 2.x here:

https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing

And how to use the autocomplete feature of Arduino IDE 2.x (which is provided by Arduino Language Server under the hood) here:

https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature

closed 2023-06-11 by per1234

ubaldot, 2023-06-11:
Usch! I installed Arduino IDE 2.x today and I couldn't figure how to trigger the autocomplete, so I installed the LSP. The guide you linked is eye opener though! Thanks! :)