arduino/arduino-language-server #166: Arduino ls crashing in neovim in wsl

CWood-sdf, 2023-09-10:
### Describe the problem

I set up arduino cli in neovim inside ubuntu wsl with lspconfig and mason.nvim. The problem is, it crashes whenever I even enter a buffer. When I look through the log, the compile command fails because it has the flag " --source-override /tmp/2700121934", removing this flag and running the compile command works from the command line. So when I tried removing that from the source code, it gave me an error of "could not find compile_commands.json"

Here's my init for the lsp:
require('lspconfig').arduino_language_server.setup({
    cmd = { "<path-to-arduino-ls>", "-clangd",
        "$HOME/.local/share/nvim/mason/bin/clangd", "-cli-config",
        "$HOME/snap/arduino-cli/41/.arduino15/arduino-cli.yaml", "-fqbn", "arduino:avr:pro", "-log", "true" },
})

Here's the compile command that it crashes on:
/snap/bin/arduino-cli --config-file $HOME/snap/arduino-cli/41/.arduino15/arduino-cli.yaml compile --fqbn arduino:avr:pro --only-compilation-database --source-override /tmp/3587927762 --build-path /tmp/arduino-language-server1532322847/fullbuild --format json <PROJECT_DIR>

### To reproduce

i just created a new sketch, setup arduino ls using the code above, and tried to enter a buffer.

### Expected behavior

I would expect it to attach to the buffer and work, but it instantly crashes

### Arduino Language Server version

I tried with a build of the current branch, the build that mason uses, and a download of the 0.7.5 release

### Arduino CLI version

0.34.1

### Operating system

Linux

### Operating system version

Ubuntu 22.04.2 LTS

### Additional context

n/a

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

BennehBoy, 2024-05-24:
Hi, did you get a resolution to this, I can't get the lsp to work with neovim on arch, doesn't attach to the buffer...

CWood-sdf, 2024-05-24:
I have the same problem on ubuntu. I can start the server, but it refuses to attach. But it doesnt crash like it did in wsl

BennehBoy, 2024-05-28:
I enabled the LSP logging on the IDE, and the only difference I can see in the init sequence is that neovim's LS init request is this:
IDE --> LS REQU initialize 1

Whereas the IDE's is:
IDE --> LS REQU initialize 0

I'm not sure is this is germane.

The LS gets no further when Neovim sends the init whereas the next output (of many) from the LS when initialised by the IDE is:
initialize 0: write-locked

BennehBoy, 2024-06-12:
Downgrade neovim to 0.9.5 there's a breaking change in neovim 0.10.x

Also, disable semanticTokens in the LSP capabilities...

I'm using kickstart so I have the following added in my config:
`capabilities = {
   TextDocument = {
     semanticTokens = vim.NIL,
   },
   workspace = {
     semanticTokens = vim.NIL,
   },
}					},
`

closed 2024-08-30 by CWood-sdf