arduino/arduino-language-server #180: Getting error when trying to open a .ino file with the arduino language server

franciscodeazeredo, 2024-03-11:
### Describe the problem

Failed to run client.before_init for settings/plugins/lspconfig
vim/shared.lua:0: Cannot deepcopy object of type userdata

stack traceback:
	vim/shared.lua: in function 'deepcopy'
	vim/shared.lua: in function 'deepcopy'
	vim/shared.lua: in function 'deepcopy'
	vim/shared.lua: in function 'deepcopy'
	vim/shared.lua: in function 'deepcopy'
	vim/shared.lua: in function 'deepcopy'
	...nvim/lazy/neoconf.nvim/lua/neoconf/plugins/lspconfig.lua:44: in function 'on_config'
	....local/share/nvim/lazy/neoconf.nvim/lua/neoconf/util.lua:47: in function <....local/share/nvim/lazy/neoconf.nvim/lua/neoconf/util.lua:46>
	[C]: in function 'func'
	...al/share/nvim/lazy/nvim-lspconfig/lua/lspconfig/util.lua:54: in function 'func'
	...al/share/nvim/lazy/nvim-lspconfig/lua/lspconfig/util.lua:54: in function <...al/share/nvim/lazy/nvim-lspconfig/lua/lspconfig/util.lua:52>
	[C]: in function 'pcall'
	...share/nvim/lazy/nvim-lspconfig/lua/lspconfig/configs.lua:207: in function 'make_config'
	...share/nvim/lazy/nvim-lspconfig/lua/lspconfig/manager.lua:203: in function 'add'
	...share/nvim/lazy/nvim-lspconfig/lua/lspconfig/manager.lua:278: in function <...share/nvim/lazy/nvim-lspconfig/lua/lspconfig/manager.lua:267>
	[C]: in function 'pcall'
	...l/share/nvim/lazy/nvim-lspconfig/lua/lspconfig/async.lua:5: in function <...l/share/nvim/lazy/nvim-lspconfig/lua/lspconfig/async.lua:4>

### To reproduce

I am using Astrovim, and installed arduino language server with mason.

### Expected behavior

For the Language server to start and work. Right now it doesn't. it can just make the syntax. But no completion

### Arduino Language Server version

0.7.6

### Arduino CLI version

arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:18Z

### Operating system

Linux

### Operating system version

Linux Mint with i3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

mroavi, 2024-05-05:
I have the same issue. This LSP message gets displayed as soon as I open the .ino file: `Client 1 quit with exit code 2 and signal 0`. This Arduino LSP was working fine a few months back.

anmoljhamb, 2024-05-08:
I may have a possible fix. For everyone who is facing this issue, I was having the exact same issue, and JUST COULDN'T FIGURE IT OUT. But finally! I have managed to fix it for myself. 

My config was just the fact that I was using NeoVim with the Arduino LSP, and the Arduino-cli to to upload sketches. But to check it out, I downloaded the official ide, and turns out that the LSP wasn't working EVEN THERE! So, I tried compiling a Hello, World program, and it showed me an error saying, "serial not found", turns out, my system-wide install of python was missing the pyserial library. Which is really simple to fix! Just try the 
`python3 -m pip install pyserial --break-system-packages`.

Note: The pipx install didn't work, and had to use the --break-system-packages instead.
Just try restarting your IDE after it, and it should work!! If it doesn't, I recommend trying out by compiling a hello world program in the official ide!

anmoljhamb, 2024-05-08:
@mroavi Let me know if my suggested fix works for you!!

mroavi, 2024-05-08:
I already had pyserial installed globally 😔. 
```
❯ python3 -m pip install pyserial --break-system-packages
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyserial in /usr/lib/python3.12/site-packages (3.5)
```
I still get 
 ```
 Client 1 quit with exit code 2 and signal 0
```
when I open an .ino file in neovim.

closed 2025-03-19 by alessio-perugini