arduino/arduino-language-server #186: Arduino LS "rpc.lua" problem with neovim

Ma94xx, 2024-06-04:
### Describe the problem

I'm trying to get the Arduino LS integrated into neovim but I'm out of answers. I'm using Mason to manage the servers, and tried to fix the version as and patched with go as described in #182.
When opening a .ino file, LspInfos tells me that 0 clients are attached, and I don't get any features of the LS.

The output of LspLog is the following:
`
[ERROR][2024-06-04 10:04:33] .../vim/lsp/rpc.lua:770	"rpc"  ".../nvim/mason/bin/arduino-language-server"  "stderr"
`

The content of my arduino-lsp config:
`
local MY_FQBN = "arduino:avr:nano"
require("lspconfig").arduino_language_server.setup({
	lspconfig.arduino_language_server.setup({
		cmd = {
			"arduino-language-server",
			"-clangd",
			"/usr/bin/clangd", -- I have also tried Mason's clangd, no use either.
			"-cli-config",
			"/home/max/arduino15/arduino-cli.yaml",
			"-cli",
			"/usr/bin/arduino-cli",
			"-fqbn",
			MY_FQBN,
		},
		on_attach = on_attach,
		capabilities = capabilities,
	}),
})
`

I'm not sure if this issue is related to #155 and I hope I wrote all the details needed. Sorry in advance if this is a duplicate!

### To reproduce

1. Installed the LS via Mason in nvim and fixed the version
2. Downloaded the ls via go as described in #182 with patching the version.
3. Installed the arduino-cli via pacman
4. Installed the cores for the atmega2560 I'm using and setup via arduino-cli 
5. Wrote and added the arduino-lspconfig

### Expected behavior

Integreation of the LS in nvim with the expected features like diagnostics and completion

### Arduino Language Server version

0.7.6

### Arduino CLI version

0.35.3

### Operating system

Linux

### Operating system version

Arch Linux

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

BennehBoy, 2024-06-12:
Downgrade neovim to 0.9.5, looks like 0.10.x has a breaking change for this LSP.

You'll need to disable symanticTokens in the capabilities too.

vlappa, 2024-09-27:
Downgrading neovim only for this LSP is not a option. Would be nice if it would be fixed.

SeanMcGoff, 2024-10-01:
Dealing with this issue as well on the same architecture (arch, neovim, lspconfig). Would the fix just be implementing the new lsp capabilities in nvim 0.10?

nicolasauler, 2024-10-27:
Hey, have we had any developments on this?
Kind of a bummer that the LSP is no longer compatible with current versions of Neovim...

kryshac, 2024-12-04:
It hasn't been fixed yet, I did [this](https://github.com/arduino/arduino-language-server/pull/199#issuecomment-2453517587) and it works ok

alessio-perugini, 2025-03-19:
Fixed in #205 . Thank you all for the support! Apologies for being so late in fixing this 🙏

closed 2025-03-19 by alessio-perugini