arduino/arduino-language-server #187: nvim-lspconfig + arduino-language-server doesn't attach to buffer

realhackcraft, 2024-06-09:
### Describe the problem

I'm using NvChad. My arduino ls doesn't crash, but doesn't attach to the buffer either.

### To reproduce

my config:

```lua
local on_attach = require("nvchad.configs.lspconfig").on_attach
local on_init = require("nvchad.configs.lspconfig").on_init
local capabilities = require("nvchad.configs.lspconfig").capabilities

require("lspconfig").arduino_language_server.setup {
  cmd = {
    "arduino-language-server",
    "-cli-config",
    "$HOME/.arduino15/arduino-cli.yaml",
    "-fqbn",
    MY_FQBN,
  },
}
```

### Expected behavior

Attach to the buffer with the .ino file

### Arduino Language Server version

0.7.6

### Arduino CLI version

arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:15:51Z

### Operating system

macOS

### Operating system version

14.5

### Additional context

[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

realhackcraft, 2024-06-09:
This is in the last lines of `:LspLog`:

```
[START][2024-06-09 12:16:08] LSP logging initiated
[ERROR][2024-06-09 12:16:08] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/Borui/.local/share/nvim/mason/bin/arduino-language-server"	"stderr"	"12:16:08.152281 clangd found at /Users/Borui/.local/share/nvim/mason/bin/clangd\n12:16:08.154789 \27[97mLS: : Initial board configuration: arduino:avr:leonardo\27[0m\n12:16:08.154969 \27[97mLS: : arduino-language-server Version: 0.7.6 Commit: 9c2f44d Date: 2024-02-06T14:12:59Z\27[0m\n12:16:08.154991 \27[97mLS: : Language server temp directory: /private/var/folders/rj/bw9lqpk15g9ckdkpv0n7nhvr0000gp/T/arduino-language-server3164570696\27[0m\n12:16:08.154994 \27[97mLS: : Language server build path: /private/var/folders/rj/bw9lqpk15g9ckdkpv0n7nhvr0000gp/T/arduino-language-server3164570696/build\27[0m\n12:16:08.154995 \27[97mLS: : Language server build sketch root: /private/var/folders/rj/bw9lqpk15g9ckdkpv0n7nhvr0000gp/T/arduino-language-server3164570696/build/sketch\27[0m\n12:16:08.154997 \27[97mLS: : Language server FULL build path: /private/var/folders/rj/bw9lqpk15g9ckdkpv0n7nhvr0000gp/T/arduino-language-server3164570696/fullbuild\27[0m\n12:16:08.157664 IN Elapsed: 921.084Âµs\n12:16:08.160606 \27[92mIDE --> LS REQU initialize 1\27[0m\n"
```

It appears to be the same as the output of the `arduino-language-server` sh command with the arguments.

realhackcraft, 2024-06-09:
```
-cli-daemon-addr string
        TCP address and port of the Arduino CLI daemon (for example: localhost:50051)
```

Does this mean that I need to spin up a arduino cli daemon myself, or is the ls going to do it?

HannHank, 2024-06-09:
I encountered a similar issue, which seems to align with the one described in #186. I resolved it by downgrading to Neovim version v0.9.5. It appears that there are breaking changes in the newer versions of Neovim that prevent the LSP client from connecting properly.

realhackcraft, 2024-06-09:
Right. Thank you!

closed 2024-06-09 by realhackcraft

fspv, 2024-06-20:
I found the problem. It is in the https://github.com/bugst/go-lsp package, which arduino LSP depends on.

nvim 0.10 passes some new capabilities to the LSP, such as `inlayHint`. The problem is that the lib mentioned above seems to be not maintained and/or updated and hence it doesn't support these new capabilities (try to search for inlayHint here https://github.com/bugst/go-lsp/blob/master/lsp_capabilities_client.go).

So the solution here will be to add new capabilities to the library mentioned above. The problem is, unfortunately, not fixable from this repo, because the initial request handling and parsing is done in the LSP lib.

Vosjedev, 2024-06-26:
what are the missing capabilities? I would like to know, because then I could try setting them to false in the lsp configuration:
```lua
-- [...]
local arduino_capabilities = vim.lsp.protocol.make_client_capabilities()
-- now here set some capabilities to false:
arduino_capabilities.unsupportedCapability = false
-- then use arduino_capabilities as the capabilities value for the lspconfig entry
-- [...]
```

fspv, 2024-06-26:
I tried setting them to false, but it didn't work. I tried to look at the neovim code and I think there is no way to say not to pass inlay hint field to the language server. So even if you set it to false, it will be sent and the language server will crash, because it doesn't expect to see such a field in the json.

But anyway, I found out that you can just use `clangd` directly with the arduino code. The trick is to create a separate `test.h` and `test.cpp` files and import them from the `.ino` file.  You can find the paths of the libraries to include in `cat build/libraries.cache` generated by the `arduino-cli compile` and place them into the `.clangd` file like this

```
CompileFlags:
  Add:
  - -I/home/fspv/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino
  - -I/home/fspv/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard
  - -I/home/fspv/Arduino/libraries/MD_Parola/src
  - -I/home/fspv/Arduino/libraries/MD_MAX72XX/src
  - -I/home/fspv/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src
```

So I personally don't need an `arduino-language-server` anymore :)

xdanep, 2024-06-29:
I have the same issue, i tried everything but nothing works
[image]

leoverde2, 2024-07-21:
This fixed it for me:

Change this:
`type ProgressToken json.RawMessage`

to this:
`type ProgressToken string`

at the end of the file jsonrpc_protocol.go in the go-lsp library that this lsp depends on.