arduino/arduino-language-server #194: Updated my system and arduino-language-server crashes upon startup

Invertisment, 2024-08-09:
### Describe the problem

I enter my NVIM and it starts the LSP server.

Almost immediately the LSP server crashes with this message in the editor:
```
Client 1 quit with exit code 2 and signal 0
``

Then I get this message in the LSP log:
```
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	"15:03:11.573161 \27[96mINIT --- : running: /usr/bin/arduino-cli --config-file /home/_/.arduino15/arduino-cli.yaml config dump --format json\27[0m\n"
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	"15:03:11.582275 \27[96mINIT --- : Arduino Data Dir -> \27[0m\n"
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	"15:03:11.582366 Panic: runtime error: invalid memory address or nil pointer dereference\n\ngoroutine 18 [running]:\nruntime/debug.Stack()\n\t/usr/lib/go/src/runtime/debug/stack.go:24 +0x5e\ngithub.com/arduino/arduino-language-server/streams.CatchAndLogPanic()\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/streams/panics.go:29 +0x7e\npanic({0xaa17c0?, 0x1114170?})\n\t/usr/lib/go/src/runtime/panic.go:770 +0x132\ngithub.com/arduino/go-paths-helper.(*Path).Clone(...)\n\t/home/_/go/pkg/mod/github.com/arduino/go-paths-helper@v1.7.0/paths.go:81\ngithub.com/arduino/go-paths-helper.(*Path).Canonical(0xc000251c10?)\n\t/home/_/go/pkg/mod/github.com/arduino/go-paths-helper@v1.7.0/paths.go:536 +0x19\ngithub.com/arduino/arduino-language-server/ls.(*INOLanguageServer).extractDataFolderFromArduinoCLI(0xc000000480, {0xc84d80, 0xc000388000})\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go:1485 +0x98d\ngithub.com/arduino/arduino-language-server/ls.(*INOLanguageServer).initializeReqFromIDE.func1()\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go:215 +0x36a\ncreated by github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).initializeReqFromIDE in goroutine 8\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go:189 +0x205\n\n"
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	"panic: runtime error: invalid memory address or nil pointer dereference [recovered]\n\tpanic: runtime error: invalid memory address or nil pointer dereference\n[signal "
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	"SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x790c99]\n\ngoroutine 18 [running]:\ngithub.com/arduino/arduino-language-server/streams.CatchAndLogPanic()\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/streams/panics.go:30 +0x194\npanic({0xaa17c0?, 0x1114170?})\n\t/usr/lib/go/src/runtime/panic.go:770 +0x132\ngithub.com/arduino/go-paths-helper.(*Path).Clone(...)\n\t/home/_/go/pkg/mod/github.com/arduino/go-paths-helper@v1.7.0/paths.go:81\ngithub.com/arduino/go-paths-helper.(*Path).Canonical(0xc000251c10?)\n\t/home/_/go/pkg/mod/github.com/arduino/go-paths-helper@v1.7.0/paths.go:536 +0x19\ngithub.com/arduino/arduino-language-server/ls.(*INOLanguageServer).extractDataFolderFromArduinoCLI(0xc000000480, {0xc84d80, 0xc000388000})\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go"
[ERROR][2024-08-09 15:03:11] .../vim/lsp/rpc.lua:734	"rpc"	"/home/_/go/bin/arduino-language-server"	"stderr"	":1485 +0x98d\ngithub.com/arduino/arduino-language-server/ls.(*INOLanguageServer).initializeReqFromIDE.func1()\n\t/home/_/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go:215 +0x36a\ncreated by github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).initializeReqFromIDE in goroutine 8\n\t/home/martin/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20240206141035-9c2f44d1ba9b/ls/ls.go:189 +0x205\n"

```

### To reproduce

It's enough to start my editor with `.ino` file.

clangd version 18.1.8

NVIM v0.9.5 (I had to downgrade from 10.1 because of this bug: https://github.com/arduino/arduino-language-server/issues/186) 

### Expected behavior

No crash.

### Arduino Language Server version

0.7.6

### Arduino CLI version

1.0.3-arch Commit: 8b6ad258e88581a9e77b2eede65aeeeeb265008c

### Operating system

Linux

### Operating system version

6.9.12-1-MANJARO

### Additional context

Note: If I upgrade to the `@latest` unstable version then it doesn't crash (`go install github.com/arduino/arduino-language-server@latest`)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

alessio-perugini, 2024-08-12:
@Invertisment :wave: thank you for taking the time to open the issue. The reason is that the latest stable doesn't target the arduino-cli v1.0.0 yet. For this reason, I suggest installing the LSP with the `latest` tag `go install github.com/arduino/arduino-language-server@latest`. 
This will be needed until we release a new version, that will target the new `v1` version of the arduino-cli.

closed 2024-08-12 by alessio-perugini

smith-isaac, 2025-03-28:
Issue seems to persist using latest version of `arduino-language-server`.

Neovim version:
```
NVIM v0.11.0
Build type: Release
LuaJIT 2.1.1741730670
```

Arduino-cli version info:
```
arduino-cli  Version: 1.2.0 Commit: 9c495211 Date: 2025-02-24T15:57:30Z
```

inols-err.log
```
Started logging.
10:14:09.247681 Language server launched with arguments:
10:14:09.247691   arg[0] = arduino-language-server
10:14:09.247698   arg[1] = -log
10:14:09.247890 ArduinoCLI config file found at /home/isaacsmith/.arduino15/arduino-cli.yaml
10:14:09.247920 arduino-cli found at /home/isaacsmith/.local/bin/arduino-cli
10:14:09.247973 clangd found at /usr/bin/clangd
10:14:09.248014 logging to /home/isaacsmith/test/inols.log
10:14:09.248722 LS: : Initial board configuration:
10:14:09.248741 LS: : arduino-language-server Version: 0.0.0-git Commit:  Date:
10:14:09.248750 LS: : Language server temp directory: /tmp/arduino-language-server78730800
10:14:09.248755 LS: : Language server build path: /tmp/arduino-language-server78730800/build
10:14:09.248761 LS: : Language server build sketch root: /tmp/arduino-language-server78730800/build/sketch
10:14:09.248796 LS: : Language server FULL build path: /tmp/arduino-language-server78730800/fullbuild
10:14:09.249155 IN Elapsed: 205.375µs
10:14:09.249377 IDE --> LS REQU initialize 1
10:14:09.250770                 initialize 1: write-locked
10:14:09.251011 Panic: runtime error: invalid memory address or nil pointer dereference

goroutine 23 [running]:
runtime/debug.Stack()
        /opt/go/src/runtime/debug/stack.go:26 +0x5e
github.com/arduino/arduino-language-server/streams.CatchAndLogPanic()
        /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/streams/panics.go:29 +0x75
panic({0xab36c0?, 0x11425d0?})
        /opt/go/src/runtime/panic.go:792 +0x132
github.com/arduino/go-paths-helper.(*Path).Clone(...)
        /home/isaacsmith/go/pkg/mod/github.com/arduino/go-paths-helper@v1.12.1/paths.go:87
github.com/arduino/go-paths-helper.(*Path).Canonical(0xc000399868?)
        /home/isaacsmith/go/pkg/mod/github.com/arduino/go-paths-helper@v1.12.1/paths.go:558 +0x19
go.bug.st/lsp.DocumentURI.AsPath({{{0x0, 0x0}, {0x0, 0x0}, 0x0, {0x0, 0x0}, {0x0, 0x0}, {0x0, ...}, ...}})
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/uri.go:42 +0x65
github.com/arduino/arduino-language-server/ls.(*INOLanguageServer).initializeReqFromIDE(0xc00015c180, {0xc0003860b0?, 0xa55980?}, {0xc8b720, 0xc0002e80a8}, 0xc00038c000)                                                                           /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/ls/ls.go:184 +0x98
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).Initialize(0xc000385000?, {0xc92608?, 0xc000290370?}, {0xc8b720?, 0xc0002e80a8?}, 0x4fa1a0?)
        /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/ls/lsp_server_ide.go:58 +0x2b
go.bug.st/lsp.(*Server).requestDispatcher(0xc0001babd0, {0xc92608, 0xc000290370}, {0xc8b720, 0xc0002e80a8}, {0xc000292260, 0xa}, {0xc000385000, 0xf1d, 0x1000}, ...)
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/server.go:275 +0x188
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingRequest(0xc00018e980, 0xc0002902d0)
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/jsonrpc/jsonrpc_connection.go:153 +0x2af
go.bug.st/lsp/jsonrpc.(*Connection).handleIncomingData(0xc00018e980, {0xc000384000, 0xf55, 0xf55})
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/jsonrpc/jsonrpc_connection.go:128 +0x167
go.bug.st/lsp/jsonrpc.(*Connection).Run(0xc00018e980)
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/jsonrpc/jsonrpc_connection.go:119 +0x54
go.bug.st/lsp.(*Server).Run(...)
        /home/isaacsmith/go/pkg/mod/go.bug.st/lsp@v0.1.3/server.go:134
github.com/arduino/arduino-language-server/ls.(*IDELSPServer).Run(...)
        /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/ls/lsp_server_ide.go:53
github.com/arduino/arduino-language-server/ls.NewINOLanguageServer.func1()
        /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/ls/ls.go:174 +0x50
created by github.com/arduino/arduino-language-server/ls.NewINOLanguageServer in goroutine 1
        /home/isaacsmith/go/pkg/mod/github.com/arduino/arduino-language-server@v0.0.0-20250319095226-9945da32e7fa/ls/ls.go:172 +0x711
```

@alessio-perugini I have tried both version 0.7.7 and the latest version of arduino-language-server, and both seem to have the same issue. Should I have to downgrade neovim or the arduino-cli versions?