arduino/arduino-language-server #202: LSP fails to bind with a Neovim Buffer

gmmyung, 2024-12-02:
### Describe the problem

I am not sure if this is a duplicate, but I cannot make `arduino-language-server` work on neovim.

LSP Log:
```
[START][2024-12-02 15:24:46] LSP logging initiated
[ERROR][2024-12-02 15:24:46] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/gmmyung/go/bin/arduino-language-server"	"stderr"	"15:24:46.322330 arduino-cli found at /opt/homebrew/bin/arduino-cli\n15:24:46.322454 clangd found at /opt/homebrew/opt/llvm/bin/clangd\n"
[ERROR][2024-12-02 15:24:46] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/gmmyung/go/bin/arduino-language-server"	"stderr"	"15:24:46.323107 \27[97mLS: : Initial board configuration: arduino:avr:nano\27[0m\n15:24:46.323114 \27[97mLS: : arduino-language-server Version: 0.0.0-git Commit:  Date: \27[0m\n"
[ERROR][2024-12-02 15:24:46] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/gmmyung/go/bin/arduino-language-server"	"stderr"	"15:24:46.323124 \27[97mLS: : Language server temp directory: /private/var/folders/pn/1x_sn6ws1qz96cvyl5jqjfxh0000gn/T/arduino-language-server3193239731\27[0m\n15:24:46.323126 \27[97mLS: : Language server build path: /private/var/folders/pn/1x_sn6ws1qz96cvyl5jqjfxh0000gn/T/arduino-language-server3193239731/build\27[0m\n15:24:46.323128 \27[97mLS: : Language server build sketch root: /private/var/folders/pn/1x_sn6ws1qz96cvyl5jqjfxh0000gn/T/arduino-language-server3193239731/build/sketch\27[0m\n15:24:46.323129 \27[97mLS: : Language server FULL build path: /private/var/folders/pn/1x_sn6ws1qz96cvyl5jqjfxh0000gn/T/arduino-language-server3193239731/fullbuild\27[0m\n"
[ERROR][2024-12-02 15:24:46] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/gmmyung/go/bin/arduino-language-server"	"stderr"	"15:24:46.323495 IN Elapsed: 266.917µs\n"
[ERROR][2024-12-02 15:24:46] .../vim/lsp/rpc.lua:770	"rpc"	"/Users/gmmyung/go/bin/arduino-language-server"	"stderr"	"15:24:46.323665 \27[92mIDE --> LS REQU initialize 1\27[0m\n"
```
Nvim version:
```
NVIM v0.10.2
Build type: Release
LuaJIT 2.1.1731601260
```

### To reproduce

Open the `ino` file buffer

### Expected behavior

The LSP should bind to the buffer

### Arduino Language Server version

0.7.6

### Arduino CLI version

arduino-cli  Version: 1.1.1 Commit: Homebrew Date: 2024-11-22T08:56:36Z

### Operating system

macOS

### Operating system version

15.1.1 (24B91)

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-language-server/issues?q=)
- [x] I verified the problem still occurs when using the latest version
- [x] My report contains all necessary details

defliez, 2024-12-06:
Hey, please check the conversation for this pull request and see if it works for you: 
#199

SaintFTS, 2025-01-09:
Same for me. Nothing works. 
It doesn't show any errors untill i manually type :LspRestart. And when i do, the error pops up:

`Client arduino_language_server quit with exit code 1 and signal 15. Check log for errors: C:\Users\Admin\AppData\Local\nvim-data/lsp.log`

Part of my init.lua (every tool has latest updates and they all work flawlessly if used separately):
```lua
local lspconfig = require("lspconfig")
lspconfig.arduino_language_server.setup({

  --filetypes = { "c", "cpp", "objc", "objcpp", "arduino" },
  enable = true,
  cmd = {
    "arduino-language-server",
    "-cli-config",
    "C:/Users/Admin/AppData/Local/Arduino15/arduino-cli.yaml",
    "-fqbn",
    "arduino:avr:uno",
    "-cli",
    "C:/Program Files/Arduino CLI/arduino-cli.exe",
    "-clangd",
    "C:/Users/Admin/AppData/Local/clangd/clangd.exe",
    --"esp32:esp32:esp32wroover",
  },
  disabledFeatures = { "semanticTokens" },
}
```
:LspLog
```
[START][2025-01-09 17:32:43] LSP logging initiated
[ERROR][2025-01-09 17:32:43] .../vim/lsp/rpc.lua:770	"rpc"	"C:\\Users\\Admin\\AppData\\Local\\nvim-data\\mason\\bin\\arduino-language-server.CMD"	"stderr"	"17:32:43.599106 \27[97mLS: : Initial board configuration: arduino:avr:uno\27[0m\n17:32:43.607657 \27[97mLS: : arduino-language-server.exe Version: 0.7.6 Commit: 9c2f44d Date: 2024-02-06T14:13:05Z\27[0m\n17:32:43.607657 \27[97mLS: : Language server temp directory: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server1879090863\27[0m\n17:32:43.607657 \27[97mLS: : Language server build path: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server1879090863\\build\27[0m\n17:32:43.607657 \27[97mLS: : Language server build sketch root: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server1879090863\\build\\sketch\27[0m\n17:32:43.607657 \27[97mLS: : Language server FULL build path: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server1879090863\\fullbuild\27[0m\n17:32:43.607657 IN Elapsed: 0s\n17:32:43.607657 \27[92mIDE --> LS REQU initialize 1\27[0m\n"
[ERROR][2025-01-09 17:32:51] .../vim/lsp/rpc.lua:770	"rpc"	"C:\\Users\\Admin\\AppData\\Local\\nvim-data\\mason\\bin\\arduino-language-server.CMD"	"stderr"	"17:32:51.261010 \27[97mLS: : Initial board configuration: arduino:avr:uno\27[0m\n17:32:51.269388 \27[97mLS: : arduino-language-server.exe Version: 0.7.6 Commit: 9c2f44d Date: 2024-02-06T14:13:05Z\27[0m\n17:32:51.269388 \27[97mLS: : Language server temp directory: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server4049435830\27[0m\n17:32:51.269388 \27[97mLS: : Language server build path: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server4049435830\\build\27[0m\n17:32:51.269388 \27[97mLS: : Language server build sketch root: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server4049435830\\build\\sketch\27[0m\n17:32:51.269388 \27[97mLS: : Language server FULL build path: C:\\Users\\Admin\\AppData\\Local\\Temp\\arduino-language-server4049435830\\fullbuild\27[0m\n17:32:51.269388 IN Elapsed: 0s\n"
[ERROR][2025-01-09 17:32:51] .../vim/lsp/rpc.lua:770	"rpc"	"C:\\Users\\Admin\\AppData\\Local\\nvim-data\\mason\\bin\\arduino-language-server.CMD"	"stderr"	"17:32:51.269388 \27[92mIDE --> LS REQU initialize 1\27[0m\n"
```

MahmoodSeoud, 2025-03-12:
I followed the fix from here: [#199](https://github.com/arduino/arduino-language-server/pull/199#issuecomment-2519818108), I was able to get the Arduino LSP working by configuring it like this:  

```lua
lspconfig.arduino_language_server.setup({
	cmd = {
		vim.fn.expand("~/.local/share/nvim/mason/bin/arduino-language-server"),
		"-clangd",
		"clangd",
		"-cli",
		"/opt/homebrew/bin/arduino-cli",
		"-cli-config",
		vim.fn.expand("~/Library/Arduino15/arduino-cli.yaml"),
		"-fqbn",
		"esp32:esp32:t-beam",
	},
	filetypes = { "arduino" },
	autostart = true,
})
```

alessio-perugini, 2025-03-19:
Fixed in https://github.com/arduino/arduino-language-server/pull/205 . Thank you all for the support! Apologies for being so late in fixing this 🙏