arduino/arduino-language-server #206: Neovim 0.10+ fix

alessio-perugini, 2025-03-19:
First of all, thank you all for being so patient and kind in providing triage and possible fixes. 
Special thanks to @leoverde2, @mike-lloyd03 for the fix, and also @speelbarrow for providing a usable branch/fork.

We have released the [0.7.7](https://github.com/arduino/arduino-language-server/releases/tag/0.7.7) that contains the fix. It is now possible to use it with nvim 0.10+.

Remember that the current LSP implementation doesn't support all the features and you must ensure that you're not passing additional capabilities to your nvim config.

Example:
```lua
require("lspconfig")["arduino_language_server"].setup({
	cmd = {
		vim.fn.expand("~/.local/share/nvim/mason/bin/arduino-language-server"),
		"-clangd",
		"clangd",
		"-cli",
		"/usr/bin/arduino-cli",
		"-cli-config",
		vim.fn.expand("~/.arduino15/arduino-cli.yaml"),
		"-fqbn",
		"arduini:avr:uno",
	},
	-- capabilities = capabilities, <-- DO NOT specify that
})
```

Hopefully, when we have more bandwidth we will release an update to https://github.com/bugst/go-lsp to support more capabilities, thus avoiding crashes.

Thank you again and apologies for the huge delay 🙏

newmanls, 2025-05-17:
Got it working this way on neovim 0.11 using the new lsp api features:

```lua
-- ~/.config/nvim/after/lsp/arduino_language_server.lua
return {
    cmd = {
        'arduino-language-server',
        '-clangd'     , 'clangd',
        '-cli'        , '/usr/bin/arduino-cli',
        '-cli-config' , vim.fn.expand('~/.arduino15/arduino-cli.yaml'),
        '-fqbn'       , 'esp32-bluepad32:esp32:esp32',
    },
}
```

Both `arduino-language-server` and `clangd` were installed using mason. This is my plugin config (using [mini.deps](https://github.com/echasnovski/mini.nvim/blob/main/readmes/mini-deps.md)):

```lua
local add, now, later = MiniDeps.add, MiniDeps.now, MiniDeps.later

now(function()
    add('mason-org/mason.nvim')
    add('mason-org/mason-lspconfig.nvim')
    add('neovim/nvim-lspconfig')

    require("mason").setup()
    require("mason-lspconfig").setup({
        ensure_installed = { 'clangd', 'arduino_language_server' }
    })
end)
```

Jvpanda, 2025-06-07:
For anybody who is trying to use the new vim.lsp api in 0.11 raw, it would probably look something like this. If you want to use the lsp directory you can, just return these values like in the example above. Note, I am setting the root dir inadequately, but it works for my purposes. If you want to set the root dir in a different way I suggest looking at the :help root_dir, :help vim.lsp.enable, or neovim/nvim-lspconfig docs for some examples. If you do not set the root dir, then you will get errors about a nil pointer to a nil rootURI in the logs and it will exit.

```lua
    vim.lsp.config("arduino_language_server", {
        capabilities = {
            textDocument = {
                semanticTokens = vim.NIL,
            },
            workspace = {
                semanticTokens = vim.NIL,
            },
        },

        cmd = {
            "arduino-language-server",
            "-cli-config",
            "path to your cli configs go here",
            "-fqbn",
            "arduino:avr:exampleText",
            "-cli",
            "arduino-cli",
            "-clangd",
            "clangd",
        },

        filetypes = { "arduino" },

        root_dir = function(bufnr, on_dir)
            on_dir(vim.fn.expand "%:p:h")
        end,
    })
```